atc1441/ATC_MiThermometer #341: Will hardware rev. B1.5 work with this exploit (soon)?

kintaro829, 2025-01-13:
My Xiaomi Sensor has hardware rev. B1.5. Will this also work with rev B1.5 in near future?

atc1441, 2025-01-13:
It should work. 
Make sure to reset (Unplug the battery once) and then activate it freshly in MiHome before doing the exploit

kintaro829, 2025-01-13:
@atc1441 Thanks. I've added the device in my Xiaomi Home App already to get the ID, BLE and Token information. Do i need a Gateway for the activation?

atc1441, 2025-01-13:
No gateway is needed.
You can follow the steps as in the video: https://youtu.be/NfZHh6wmTp8

kintaro829, 2025-01-13:
I've done it already 3 times. Exact the same way you did it.

1. Add the sensor in Xiaomi Home 
2. Get the ID, BLE and Token with the Xiaomi Token Extractor
3. open your website (https://atc1441.github.io/TelinkFlasher.html)
4. Connect > select the sensor
5. input ID, BLE and Token information i've received from the Xiaomi Token Extractor
6. Login > it works
7. Select Firmware 'ATC_Exploit.bin'
8. Pressing 'Start Flashing'
9. After writing 4355 blocks the device didnt reboot like in your video (is this the problem?)

atc1441, 2025-01-13:
Ok then please reflash the stock firmware first: https://github.com/user-attachments/files/18345799/2.1.1_0159_with_cert.zip
With the same procedure and afterwards do the exploit flash

atc1441, 2025-01-13:
What Firmware version is your Thermometer showing in the TelinkFlashter?

kintaro829, 2025-01-13:
@atc1441 
17:39:14: Searching for devices
17:39:43: Connecting to: LYWSD03MMC
17:39:48: Detected Mi Thermometer
17:39:48: Fw_version: 2.1.1_0159
17:39:48: Connected

I try to reflash the firmware and do the same procedure again.

kintaro829, 2025-01-13:
@atc1441 

Nice. This worked for me. Thanks :)

atc1441, 2025-01-13:
Perfect. thanks for the feedback, it is indeed a different version which does need a different method.
So first flashing that is the solution for now
Added a warning in the flasher:
[image]

kintaro829, 2025-01-13:
@atc1441 
Thank you very much for the help :)
This will help a lot of other users, too.

frzme, 2025-01-13:
My device arrived today. I've tried flashing the original firmware 2.1.1_0159 you provided multiple times. When doing that the device reboots.

No luck whatsoever to get it to visibly reboot when flashing the ATC_Exploit firmware. After attempting to flash the firmware reconnect seems to be no longer possible and the device also is not found in the device scan dialog anymore until taking out the battery.

How can I learn about the HW revision I have?
If it helps and you are interested in looking into it I can send you the device.

atc1441, 2025-01-13:
@frzme Hard to tell what exactly is wrong. You can read the Hardware details via the nRFConnect App and share that. But right now no hardware variante should be incompatible.

atc1441, 2025-01-13:
Please send me a mail @frzme  we can discuss sending one further there: atc1441@gmail.com

VoronovAlexander, 2025-01-14:
@atc1441 Thanks you very much for your job!

I tried exploit on my two LYWSD03MMC (CMIT ID: 2019DP8115, SKU: NUN4106CN, date on boxes: 2024.08, stock firmware: 2.1.1_0159.

My worked alghoritm:
1. Unpack new sensor, add to Mi home, terminate app on phone
2. Extract tokens with https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor
3. Flash on https://github.com/atc1441/ATC_MiThermometer/issues/341#issuecomment-2587619701 with https://atc1441.github.io/TelinkFlasher.html with tokens by previous step.  - I see restarting sensor.
4. Reconnect. Repeat flash with ATC Exploit, uses tokens from step 2. - Now I see restarting sensor like on YouTube video.
5. Reconnect. Repeat flash with ATC_v49 firmware without using tokens
Then I repeated for second sensor without any problems)

For info: in first attempt I tried flash stock 2.1.1_0159 from README.md (https://github.com/atc1441/ATC_MiThermometer/blob/master/Stock_fw_2.1.1_0159.bin), **but its not working for me** - sensor freezes and does not respond when I flashed exploit, after restart restored stock firmware and continued work.

atc1441, 2025-01-14:
Thank you for that! It all makes sense now why people have problems... the stock firmware in the repo was without the signing so it was never flashing to the "good" stock one :) Will update the repo stock firmware to the one with signing now.

frzme, 2025-01-14:
Thank you! It now worked with the updated/signed original firmware. With that I'm happy and everything works as expected. Thanks a lot for the great work!

tueboesen, 2025-01-21:
This is likely a dumb question, but I am stuck getting the devices succesfully into the MiHome homepage. I downloaded the mihome app and found the sensor in there and added it. However the devices does not show up online in my xiaomi account, and when I try to use the token extractor it is able to login to my account but it does not find any devices.

In the app I see a message about no bluetooth gateway connected, is this the issue? and do I need a separate device for this or is this just some misconfiguration in the app?

Sorry if this isn't the right place to ask this question, but I could not find a more appropriate place.