atc1441/ATC_MiThermometer #342: Thermometer flashes successfully, but never comes back with custom firmware

slawrensen, 2025-01-14:
I followed the steps exactly per the video and the flashes always work, but for some reason the device never ends up with custom firmware, and doesn't report having custom firmware when reconnecting with your flasher.

I flashed the stock firmware provided by you, then the exploit, but it's still coming back as if it has stock firmware and I can't figure out why. I even flashed pvvx's firmware after which appeared succesful, but alas it still appears to have stock firmware!

18:18:54: Received device infos are correct
18:18:54: Login successfull
18:18:58: File was selected, size: 104404 bytes
18:18:58: Count: 6526
18:18:58: Start DFU
18:20:17: Update done after 77.82 seconds
18:20:24: Disconnected.
20:38:34: Reconnect
20:38:43: Detected Mi Thermometer
20:38:43: Fw_version: 2.1.1_0159
20:38:43: Connected
20:38:51: File was selected, size: 69680 bytes
20:38:51: Count: 4355
20:38:56: Start DFU
20:38:56: Please do the Activation first
20:39:14: Received device infos are correct
20:39:14: Login successfull
20:39:21: Start DFU
20:40:16: Update done after 54.16 seconds
20:40:22: Disconnected.
20:40:55: Reconnect
20:41:03: NetworkError: Connection Error: Connection attempt failed.
20:41:03: Reconnect 1 from 5
20:41:11: NetworkError: Connection Error: Connection attempt failed.
20:41:11: Reconnect 2 from 5
20:41:18: NetworkError: Connection Error: Connection attempt failed.
20:41:18: Reconnect 3 from 5
20:41:22: Searching for devices
20:41:26: NetworkError: Connection Error: Connection attempt failed.
20:41:26: Reconnect 1 from 5
20:41:35: NetworkError: Connection Error: Connection attempt failed.
20:41:35: Reconnect 2 from 5
20:41:39: NotFoundError: User cancelled the requestDevice() chooser.
20:41:39: Reconnect 3 from 5
20:41:43: NetworkError: Connection Error: Connection attempt failed.
20:41:43: Reconnect 4 from 5
20:41:43: NetworkError: Connection Error: Connection attempt failed.
20:41:43: Reconnect 5 from 5
20:41:50: NetworkError: Connection Error: Connection attempt failed.
20:41:50: Something went wrong, to many reconnect's
20:41:50: NetworkError: Connection Error: Connection attempt failed.
20:41:50: Reconnect 1 from 5
20:41:58: NetworkError: Connection Error: Connection attempt failed.
20:41:58: Reconnect 2 from 5
20:42:06: NetworkError: Connection Error: Connection attempt failed.
20:42:06: Reconnect 3 from 5
20:42:14: NetworkError: Connection Error: Connection attempt failed.
20:42:14: Reconnect 4 from 5
20:42:21: NetworkError: Connection Error: Connection attempt failed.
20:42:21: Reconnect 5 from 5
20:42:29: NetworkError: Connection Error: Connection attempt failed.
20:42:29: Something went wrong, to many reconnect's
20:44:43: Searching for devices
20:44:49: Connecting to: LYWSD03MMC
20:45:01: Disconnected.
20:45:01: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
20:45:01: Reconnect 1 from 5
20:45:37: Disconnected.
20:45:37: Searching for devices
20:45:40: Connecting to: LYWSD03MMC
20:45:40: Disconnected.
20:45:40: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
20:45:40: Reconnect 1 from 5
20:45:40: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
20:45:40: Reconnect 2 from 5
20:45:53: Detected Mi Thermometer
20:45:53: Detected Mi Thermometer
20:45:53: NetworkError: GATT operation already in progress.
20:45:53: Reconnect 3 from 5
20:45:53: Detected Mi Thermometer
20:45:53: NetworkError: GATT operation already in progress.
20:45:53: Reconnect 4 from 5
20:45:53: Detected Mi Thermometer
20:45:53: NetworkError: GATT operation already in progress.
20:45:53: Reconnect 5 from 5
20:45:53: Detected Mi Thermometer
20:45:53: NetworkError: GATT operation already in progress.
20:45:53: Something went wrong, to many reconnect's
20:45:53: Fw_version: 2.1.1_0159
20:45:53: Connected

slawrensen, 2025-01-14:
I should add that when flashing the exploit, it never reboots with "ATC" on the screen as expected. I can leave it for a while and manually reboot, and it still has stock firmware in the end.

atc1441, 2025-01-14:
Hey, 
Please share the hardware details of your device.
Pictures and also the nRFConnect details as in the following picture
[image]

We will then see further.

Also you can try out Victors flasher:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

Select the ATC_Thermometer instead the exploit when flashing with that one:
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer.bin

Please report back :)

slawrensen, 2025-01-14:
[image]
[image]

I will try out Victor's fw and flasher, thx!

atc1441, 2025-01-14:
Thank you. From the infos it should all work without problems
Maybe also try to flash with a different device like smartphone or Pc just to see if it helps

slawrensen, 2025-01-14:
Same results on another PC. Also unsucessful with Victor's firmware and flasher. I do note with his flash, the LCD does change state, but doesn't seem to reboot. After resetting battery, still has stock firmware.

atc1441, 2025-01-14:
Ok thanks for testing 
Might seem like the first flash of the other stock firmware already fails as then the descripted behavior occours

Please wait a while before trying to flash again as there is a minimal risk of bricking depending on what the firmware executes with the exploit.
Flashing via UART will allways work

slawrensen, 2025-01-14:
I will try UART later as I originally intended

When flashing the stock firmware you provided, it does appear like the unit reboots each time. Screen goes off, then comes back on. With any of the "exploit" firmwares it doesn't appear that a reboot actually occurs, but with Victor's firmware it "freezes" on the screen where all LCD elements are filled

atc1441, 2025-01-14:
The stock firmwarw reboot will always happen no matter if gone good or bad. And since that naming is the same to the old stock firmware its hard to tell if it successful

Thank you for the feedback. Please keep one unflashed so we can later see if a fix will solve your flashing problem

slawrensen, 2025-01-14:
Will do. Appreciate your hard work and tinkering to figure out these fun projects!

atc1441, 2025-01-14:
Hey again, turns out I did a wrong upload on the Stock firmware in the repo and this explains why it was never fully flashing. I just now added the correct one so please redownload the stock firmware and follow the procedure :)

slawrensen, 2025-01-14:
It works! Both are flashed successfully, thanks again!

closed 2025-01-14 by slawrensen