atc1441/ATC_MiThermometer #343: Can't unbrick LYWSD03MMC by wires B1.5

VoronovAlexander, 2025-01-14:
Hi !
I bricked my sensor LYWSD03MMC (CMIT ID: 2019DP8115, SKU: NUN4106CN, B1.5, date on boxes: 2024.08) a few days ago:
when I enable power - I see only super low light on display,
when i disable power - it disappears.
I thinks somethins anyways is works.

Today second day how I have been try flash sensor by wires.

Earlier I already flashes 3 sensors by programmer for esp01: but I noticed some peculiarities - i need very a lot of attempts for good flash, devices bricks, by 20-30 try of flash its works!
I use MacOS, and have earlier one success flash by web app and two success flash by ATCtelink.py.
But now is not the case.

Also I have some tens attempts flash by https://sl.aliexpress.ru/p?key=XyKhGzL - failed.

Many attempts flash by https://sl.aliexpress.ru/p?key=gCKhGWJ - like here -> https://www.manelrodero.com/blog/flasheo-de-xiaomi-lywsd03mmc-con-firmware-2110159, but its too failed.

Also I have attempt flash from Windows with web app with FT232RL/CP2102, by its too failed!

Now I try flash by CP2102, I try "activate" with connection P14 with "+" when starting activating before flash on web app, I even connect Reset to DTR, but all its failed.

I tried different methods connection:
1. tx -> p14, without rx
2. tx -> 1kOhm -> p14, without rx
3. tx -> 1kOhm -> p14, rx -> p14

I see that ATCtelink.py works only with 3rd method.

Please help me, how I can check for my chip is fine? Maybe I killed him? For testing i have osciloscope but I don't know how to use it, maybe its will help me?
Why i have these problems with need many attempts for flash?

[image]
[image]

atc1441, 2025-01-14:
Hey. It all looks good except that you use MAC to flash. Only windows is supported.
You can use VM/emulated Windows in a MAC

atc1441, 2025-01-14:
The ship will not be broken, its just that the flashing method is not done correctly, the reason is hard to tell

VoronovAlexander, 2025-01-24:
Hi! In finally now I flashed this sensor!

I used
- macbook m1 pro macOS (15.1.1)
- unusual flasher: arduino uno r3 (ttl chip CH340G) with wire between RST and GND (I saw info that arduino uno has a physical connection of ch340 to RX/TX pins, at the same time, the wire between RST and GND stops running arduino, in the end i used only ch340g from board) 
- resistor 1kOhm between Arduino TX and RX,
- wire from Arduino RX to p14 (green wire on photo)
- 3v3 from Arduino to sensor
- GND from Arduino to sensor
- https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
with next settings: baud: 921600, ActivationTime: 1 sec

I did a lot failed attempts to flash with ATCtelink.py at different baud rates and different tacts before it was successful from the web interface.

I flashed from the web interface 2 times in a row, actually I pressed the "flash" button two times, without disconnecting the board or anything else, and on the second try everything was successful!

After the successful flashing I set a pincode and after disconnecting I could not connect to the sensor via bluetooth anymore.

I pressed «Erase all flash» three-four times in a row and rebooted a sensor: I saw result as at a beginning this way: very weak light on a display.
Then I flashed again two times and sensor worked again!

Maybe just I lucked two times)

GitHub-Noub, 2025-02-07:
FWIW, I recently had very similar symptoms on a couple of the same devices and noted these observations:

1. **Blank screen caused by misfitting LCD to circuit board connection.** There are two small 'pegs' which hold the board in position. On a couple of devices I was flashing I had replaced the board and then pushed home the cover but had picked the device up to push the cover home. Inadvertently by picking the device up without the backing being fully pushed home first the board slipped off the peg and by pushing the cover home I crushed one of the pegs - this caused intermittent connection issues with the LCD. Ensuring the cover was replaced only when the device was flat overcame this issue.
 
2. **Flashing using a CP2102 USB to TTL module** - the device would appear to flash correctly but I noted unless the blue led on the CP2102 flickered (to denote data transfer), the device wouldn't flash - in fact in one case the blue light would only flicker when I lifted the data connection from the circuit board and it stopped if I reconnected. This may be use error (connectivity issues) on my part! as I'm only just getting the hang of this TTL process.

3. **Repeated flashing** would eventually allow the device to start up with the LCD displaying correctly.

4. A device that appeared to be 'bricked' would still connect and record valid temp/humidity in the app so was to all intents and purposes still working despite having a blank LCD.