atc1441/ATC_MiThermometer #346: MIThermometer is blank screen after a try - B1.5 - 2.1.1_0159

speedy-moe, 2025-01-31:
Hi,
i cannot flash my MIThermometer back to life.
It ends always by a blank screen.

My hardware is this

[image]

https://de.aliexpress.com/item/1005004621529953.html

and 

[image]

https://de.aliexpress.com/item/1005007416816060.html

Can anyone explain me the right way to exploit/flash this Thermometer with B1.5 - 2.1.1_0159 Firmware please?

Cisor, 2025-03-10:
I have the same blank screen issue

* Added to Xiaomi app
* Captured credentials with token_extractor
* Connected in Telink FLasher
* Login with credentials
* Flashed Exploit
* Set settings
* Saved settings
* Blank screen

smstuff, 2025-06-06:
Looks like a duplicate ticket (https://github.com/atc1441/ATC_MiThermometer/issues/350) has been opened.