atc1441/ATC_MiThermometer #350: Blank screen but bluetooth available after flashing

RWrightWells, 2025-03-30:
I thought I had a brick as the screen was blank but I could still see connect via bluetooth, 

1. created Xiaomi account and added device to Xiaomi app
2.  used https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor to obtain BLE key and Token.
3.  downloaded Stock_fw_2.1.1_0159, ATC_Expliot and ATC_Thermometer.
4. Connected with bluetooth Xiamoi named device
5. Added BLE Key , Token etc and Loggin in successfully
6. Flashed with ATC_Exploit.
7. Power cycled device , blank screen
9. Connected with ATC bluetooth id
10. Flashed Stock_fw_2.1.1_0159
11.  Screen working
12. added device again to Xiaomi app
13. used extractor to obtain BLE key and Token.
14. Connected with bluetooth Xiamoi named device
15. Added BLE Key , Token etc and Loggin in successfully
16. Flashed with ATC_Exploit.
17. Flashed with ATC_Thermometer.
18. Power cycled device , blank screen.

Is there an issue with the new devices being shipped ot have I done something wrong?

Chanabra, 2025-03-31:
[surprised]     Brandon Norman reacted to your message:

  8.  Connected with ATC bluetooth id
  9.  Flashed Stock_fw_2.1.1_0159
  10. Screen working
  11. added device again to Xiaomi app
  12. used extractor to obtain BLE key and Token.
  13. Connected with bluetooth Xiamoi named device
  14. Added BLE Key , Token etc and Loggin in successfully
  15. Flashed with ATC_Exploit.
  16. Flashed with ATC_Thermometer.
  17. Power cycled device , blank screen.

  8.  Connected with ATC bluetooth id
  9.  Flashed Stock_fw_2.1.1_0159
  10. Screen working
  11. added device again to Xiaomi app
  12. used extractor to obtain BLE key and Token.
  13. Connected with bluetooth Xiamoi named device
  14. Added BLE Key , Token etc and Loggin in successfully
  15. Flashed with ATC_Exploit.
  16. Flashed with ATC_Thermometer.
  17. Power cycled device , blank screen.

atc1441, 2025-03-31:
Hey @RWrightWells there are some PCB versions which doe result in this. They use a different LCD Controller.
It is advised to the Victors version of the ATC_MiThermometer as it is more advanced, I stopped implementing further hardware revisions as it would be just the work done twice that victor already thankfully did
https://github.com/pvvx/ATC_MiThermometer

RWrightWells, 2025-03-31:
Great thanks, just saw a youtube comment you had made saying the same thing so I'm currently down that rabbit hole :)

RonandoDeLaVega, 2025-06-06:
Hello @RWrightWells, by any chance, did you get this issue solved ?
Thanks !

apppie123, 2025-06-14:
> Hey [@RWrightWells](https://github.com/RWrightWells) there are some PCB versions which doe result in this. They use a different LCD Controller. It is advised to the Victors version of the [...]

There are some open issues(settings getting lost at empty battery and screen not working at 2025.03 batch) and it appears victor is very busy at the moment since he doesn't respond to these anymore. I noticed the readme has changed that he will also stop with custom formats and only supports BTHome going forward. It's very unfortunate. So I hope this project could be revived somehow.