atc1441/ATC_MiThermometer #357: B1.6 - After flashing still old firmware inside

Easyrunner, 2025-07-30:
Hi there,

I have two LYWSD03MMC where after "successfull" flashing still the old firmware is in.

They have this Fw_version: 2.1.1_0159 flashed.

First I add the devices into my Xiaomi account.
Then I use the token extractor to get the ID, token and key which I inserd at Telink flasher after connection.

It says Login successfull.

Then I load the needed "ATC_Exploit.bin" and start flashing. Flasing shows also successfull, but nothing happen. No "ATC..." in the Screen like other devices.
It has still the old original firmwar flashed.

Here the Log:

`Log:
20:25:52: Searching for devices
20:26:12: Connecting to: LYWSD03MMC
20:26:18: Detected Mi Thermometer
20:26:18: Fw_version: 2.1.1_0159
20:26:18: Connected
20:26:58: Received device infos are correct
20:26:58: Login successfull
20:27:06: File was selected, size: 69680 bytes
20:27:06: Count: 4355
20:27:08: Start DFU
20:28:16: Update done after 66.403 seconds
20:28:22: Disconnected.`

I tried it sveral times. Always the same.

Easyrunner, 2025-09-13:
> These are the steps to update the sensors:
> 
> * Flash the two firmwares via the [TeLink Flasher](https://atc1441.github.io/TelinkFlasher.html):
>   
>   * [Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin.zip](https://github.com/user-attachments/files/22285080/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin.zip)
>   * ATC_Exploit.bin

That I can do, but after that, I don't know what to do, because I don't want Zigbee. I just want it at BLE ATC

After repowering the sensor I tried with TelinkMiFlasher the ATC_V53.bin. I can flash it but after repowering, no ATC flashed. Just the old ones.

CodeStrix, 2025-09-13:
> > > I got the sensors running.
> > > These are the steps to update the sensors:
> > > 
> > > * Flash the two firmwares via the [TeLink Flasher](https://atc1441.github.io/TelinkFlasher.html):
> > >   
> > >   * [Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin.zip](https://github.com/user-attachments/files/22285080/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin.zip)
> > >   * ATC_Exploit.bin
> > > * Then use the [TelkinMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) and flash the three firmware as described here: [Firmware via OTA for version 2.1.1_0159 pvvx/ATC_MiThermometer#602 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/602#issuecomment-2590913442)
> > > 
> > > I have 8 sensors running the zigbee firmware which are integrated into homeassistant.
> > 
> > 
> > This unfortunately does not work for me. When I try to load the ATC_Exploit.bin in the TeLink Flasher I get this error: "Select file is no telink firmware .bin". This the prevents me to propably do the following steps. I used this bin: https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Exploit.bin
> > Any thoughts?
> 
> Thats really odd. The flasher is the right one for the exploit and the firmware you referenced is the same that I have used to flash my sensors. Did you check the file?
> 
> ```
> ❯ shasum ATC_Exploit.bin
> ae8608c58a3c9a38db1f3f94adbbf54ad95e20c1  ATC_Exploit.bin
> ```

I'll check for the file when I get back from vscation. As further hints: ordered the devices 2 months ago. Tried several devices to Flash.

timbrd, 2025-09-24:
Btw: You used sha256 to calculate the checksum, I have used sha1.

This is the correct sha256 checksum:
```
❯ shasum -a 256 ATC_Exploit.bin
edcdff86ebe8251a8d61bd59d53a4ccc491333e82b31a59877dffa62de5d3b33  ATC_Exploit.bin
´´´

ehafizoglu, 2026-02-03:
Same problem.  Is there anybody that can solve this problem?